cmsis-svd/cmsis-svd #6: Some of the STMicro SVDs are named .svd

porkybrain, 2015-09-14:
https://github.com/posborne/cmsis-svd/tree/master/data/STMicro some of the STMicro SVDs are named .svd, all other files are names *.xml, is there a reason for the .svd ending rather than mxl or should I change them all?

kblomqvist, 2015-09-14:
`.svd` is an extension what ARM use.

See for example, http://www.keil.com/pack/doc/CMSIS/SVD/html/svd__example_pg.html
``

Additionally all SVD files provided via CMSIS-SVD download page are `.svd`:
http://www.keil.com/pack/doc/CMSIS/SVD/html/svd_web_public_pg.html

porkybrain, 2015-09-14:
Ok so they are really interchangeable. Any reason not to just rename them into xml? For uniformity sake.

kblomqvist, 2015-09-14:
For uniformity sake you can choose `.xml` if you like. I would go with the file extension specified by ARM.

posborne, 2015-09-15:
For now, I think renaming to .XML for consistency makes sense.  I'm guessing there is no strong preference from ARM either.  CC @salkinium @0xc0170.

0xc0170, 2015-09-15:
Agree with consistency @posborne .

I would go with .svd as pointed out earlier. However we can see most of the files in this repo are xml. I'll try to find out more, will add any update here.

porkybrain, 2015-09-15:
If they are all officially .svd maybe we should rename all the other ones.

posborne, 2015-09-15:
I'm amenable to that suggestion.  I can reopen, adjust the tests, and rename later tonight.

porkybrain, 2015-09-15:
I am for  .svd, all the files I found  were originally  .svd too, that seems to be what everyone else is naming them.

posborne, 2015-09-16:
We are now 100% .svd

commit 77505e47 (2015-09-15): cleanup: rename *.svd files to *.xml for consistence

This resolves #6
closed 2015-09-16 by posborne