cmsis-svd/cmsis-svd #8: svd files do not have enough info

porkybrain, 2015-09-16:
The bit fields of a Register can have odd behavior. An example is a bit acting as a hardware flag where the hardware can set it and the software can clear it either by reading it, writing a 1 or in some rare cases writing a 0. Another example is a bitfield acting as the end of a fifo. 

This data is however highly relevant if we are going to get serious about generating any kind of validation mechanisms from the files. 

Should we add tags to the svd files or define a svd enhancement format or something?
How responsive is ARM going to be about extending the format?
@0xC0170 what do you think?

salkinium, 2015-09-16:
Interestingly the [SVD XML Schema](http://www.keil.com/pack/doc/CMSIS/SVD/html/group__schema__1__2__gr.html) does describe a mechanism for specifying write behaviours:
```xml
 <!-- modifiedWriteValuesType specifies the pre-defined tokens for the write side effects -->
  <xs:simpleType name="modifiedWriteValuesType">
    <xs:restriction base="xs:token">
      <xs:enumeration value="oneToClear"/>
      <xs:enumeration value="oneToSet"/>
      <xs:enumeration value="oneToToggle"/>
      <xs:enumeration value="zeroToClear"/>
      <xs:enumeration value="zeroToSet"/>
      <xs:enumeration value="zeroToToggle"/>
      <xs:enumeration value="clear"/>
      <xs:enumeration value="set"/>
      <xs:enumeration value="modify"/>
    </xs:restriction>
  </xs:simpleType>
```
As far as I can tell, this field is entirely optional though and a search through the repository

So there is no need to extend the format, but unfortunately, not a single vendor added [this information to their SVD files](https://github.com/posborne/cmsis-svd/search?utf8=✓&q=modifiedWriteValues).

0xc0170, 2015-09-16:
Looks like that :/ We need to speak to vendors to get this in.

porkybrain, 2015-09-16:
Awesome that there is already a format at least. Maybe if we give them a reason the vendors will put in some effort.

porkybrain, 2015-09-17:
Looks like there is no pushToFIFO or popFromFIFO tag though. Any ideas?

salkinium, 2015-09-17:
Can you clarify what you mean by that behaviour? Perhaps give an example from a specific device datasheet?

porkybrain, 2015-09-17:
For example http://www.nxp.com/documents/user_manual/UM10732.pdf page 162 registers RBR or THR. 

In the svd https://github.com/posborne/cmsis-svd/blob/master/data/NXP/LPC11Uxx_v7.svd#L708 they are using <readAction>modify</readAction> which is better than nothing but it seems to me that the end of a FIFO should be distinguishable from say the LSR register which has the same attribute. 

I would like to be able to distinguish the end of a FIFO from other stuff so that I can make the user call 
`apply(push(THR,42));`
or 
`auto in = apply(pop(RBR));`
rather than read/write. This would make code review easier and improve static checking. 

In the future I would like to make a factory for these registers which returns a proxy which fulfills a range concept:

    copy(
        makeInputRange(RBR,read(RDR)), 
        backInserter(myInputVector)) //copy untill read(RDR) returns false

in which case it would be even more important that I can validate the first parameter passed to `makeInputRange` to be a FIFO end.

salkinium, 2015-09-17:
Hm, interesting concept.
Perhaps you can add a heuristic to the parser (as a plug-in, please!) that recognises such patterns and adds these higher-level concepts for your use.

porkybrain, 2015-09-18:
I think an extension file would be a good idea so that we can add attributes to the existing SVDs without messing with the verdor version. Most of the attributes we need to add are stuff that is standardized but left out by the vendor like `modifiedWriteValuesType`. 

Maybe two extension files one with stuff the vendor left out and one with attributes not (yet) defined by CMSIS like FIFO ends and similar.

I'm open to any other suggestions too.

porkybrain, 2015-09-20:
I'm playing around with a JSON file containing extra information about registers indexed by peripheralName::registerName with which one can add attributes to a register. The idea is that while the user walks the tree of peripherals and registers they can optionally query for more info from the JSON file. The question is where these files and there format description should live, here or in Kvasir.

posborne, 2015-09-21:
Something like a JSON file would be nice so that we can separate the metadata from individual parser implementations.  Hopefully, the code could be designed to accept both additional metadata that is in tree and external (as some pieces of information may not make sense outside of specific contexts).

porkybrain, 2015-10-02:
heres what I cam up with so far:
```json
{
	"device":{
		"cpu":{
			"name":"insert_name_here"
		}
	},
	"ADC0":{
		"interrupt":{
			"name":"aramgeddon",
			"value":63
		},
		"register":{
			"SC1A":{
				"field":{
					"ADCH":{
						"enum":{
							"000":{
								".rename":"better_name_here"
							}
						}
					}					
				}
			}
		}
	}
}
```

essentially it maps to the same property tree as the xml file, however it should be much easier to edit or write. If a tree begins with device then it is starting at the base, if it begins with a .name is assumed to be a peripheral name (with the dor removed) (this should save a lot of typing). All other names at the top level are vender/user specific stuff as stated in the CMSIS-SVD specs.

If you want to extend the vendor SVD file then just add tags (as is the example we added the CPU name and the ADCs interrupt). The only corner case is when you want to change the name of a tag because the tags are indexed by name. For this puropse there is the .rename tag with which you can change the value of an existing name. This is probably going to be most often used in conjunction with enums.

What do you guys think? There is a proof of concept already checked into Kvasir https://github.com/kvasir-io/Kvasir/tree/master/Tools/ChipFromSvdGenerator/Extention/Freescale which allows the code generator to specialize some traits depending on the included chip file to allow sexier code like this:
```C++
constexpr auto led = makePinLocation(Io::portB, Io::pin18);
constexpr auto button = makePinLocation(Io::portA, Io::pin4);

int main()
{
	apply(set(SimScgc5::portb, SimScgc5::porta));
	apply(write(PortbPcr18::mux, PortbPcr18::muxValC::v001),
		write(PortaPcr4::mux, PortaPcr4::muxValC::v001),
		makeOutput(led),
		makeInput(button));

	for(;;)
	{
		if (apply(read(button))) {
			apply(set(led));
		}
		Delay();
		apply(clear(led));
		Delay();
	}

	return 0;
}
```
the ugly enum values like `PortbPcr18::muxValC::v001` could also be fixed using the json file.