cmsis-svd/cmsis-svd #10: Inconsistant bitfield syntax bitRange vs. bitOffset and bitWidth

porkybrain, 2015-09-17:
Some files like https://github.com/posborne/cmsis-svd/blob/master/data/NXP/LPC11Uxx_v7.svd#L82 use `<bitRange>[high:low]<bitRange>` while others like https://github.com/posborne/cmsis-svd/blob/master/data/STMicro/STM32F030.svd#L60 use `<bitOffset>0</bitOffset> <bitWidth>8</bitWidth>`

It looks like the current parser only supports bitOffset and bitWidth, or this may be a false diagnosis due to my python skill still sucking.

kblomqvist, 2015-09-17:
If you take a look into CMSIS-SVD reference, there are three different ways of specify field's bit-range. http://www.keil.com/pack/doc/CMSIS/SVD/html/group__svd__xml__fields__gr.html. I have no idea why ARM has decided to provide options. It just makes parsing harder and cumbersome.

Actually in addition to this there are plenty of things, which this parser doesn't support. For example, many elements, like size, can be specified in parent elements. So, for example, if the register doesn't explicitly specify the size the parser doesn't check the value from the parent element. Also cluster elements aren't parsed at all. These are the examples I had when I was trying to use the parse for NRF51 svd-file. Eventually I decided to write my own parser, which you can find here https://github.com/kblomqvist/nrf51.rs. Instead of that everybody is writing their own parser I would like to have one fully working one, but so far I think that the parser made here would need some rethinking so I wasn't able to contribute directly as a code.

porkybrain, 2015-09-17:
One that works for everything is of course the goal, I'll have a look. I am quite committed to getting this working for all cortex m

posborne, 2015-09-18:
Good info @kblomqvist.  Hopefully we can distill that info down into some failing test cases and go from there.

porkybrain, 2015-09-18:
this seems to be working as of #12 (im running a script which generates c++ code for all svd files and no obvious errors) so I'm closing

closed 2015-09-18 by porkybrain