cmsis-svd/cmsis-svd #11: NXP seems to like naming bitfields "RESERVED"

porkybrain, 2015-09-18:
here https://github.com/posborne/cmsis-svd/blob/9f2ca607848e5b368f89e0888a0e13c7b43cf764/data/NXP/LPC15xx_v0.7.svd for example there are many registers which contain a bitfields named RESERVED. These need to be discared (especially if two RESERVED fields occure in one register) somewhere in the chain of processing. Where would it be best to di it? 

In my opinion its NXP's error so they should fix it but thats probably not going to happen so should we filter them out while parsing or should I do it in my Kvasir specific parser?

kblomqvist, 2015-09-18:
I think they use these for padding... to align structure members to their "natural" address boundaries in C.

For example:

```
typedef struct {                             
  __IO uint8_t   B[76];                      
  __I  uint32_t  RESERVED0[45];
  __IO uint32_t  W[76];                      
  __I  uint32_t  RESERVED1[1908];
  __IO uint32_t  DIR[3];                     
  __I  uint32_t  RESERVED2[29];
  __IO uint32_t  MASK[3];                    
  __I  uint32_t  RESERVED3[29];
  __IO uint32_t  PIN[3];                     
  __I  uint32_t  RESERVED4[29];
  __IO uint32_t  MPIN[3];                    
  __I  uint32_t  RESERVED5[29];
  __IO uint32_t  SET[3];                     
  __I  uint32_t  RESERVED6[29];
  __O  uint32_t  CLR[3];                     
  __I  uint32_t  RESERVED7[29];
  __O  uint32_t  NOT[3];                     
} LPC_GPIO_PORT_Type;
```

porkybrain, 2015-09-18:
there seem to be multiple cases, @kblomqvist you case is where whole registers are named reserved. I guess the RESERVED bitfields could be usefull when generating unions or something but only if NXPs parser sucks ;)

0xc0170, 2015-09-18:
@kblomqvist is right with padding. It's common , I have seen in various cmsis vendors header files.

What's NXP error?

porkybrain, 2015-09-19:
<register>
			<dim>18</dim>
			<dimIncrement>0x10</dimIncrement>
			<dimIndex>0-17</dimIndex>
			<name>CFG%s</name>
			<description>Configuration register for DMA channel 0.</description>
			<addressOffset>0x400</addressOffset>
			<access>read-write</access>
			<resetValue>0</resetValue>
			<resetMask>0x00000000</resetMask>
			<fields>
				<field>
					<name>PERIPHREQEN</name>
					<description>Peripheral request Enable. If a DMA channel is used to perform a memory-to-memory move, any peripheral DMA request associated with that channel can be disabled to prevent any interaction between the peripheral and the DMA controller.</description>
					<bitRange>[0:0]</bitRange>
					<enumeratedValues>
					<name>ENUM</name>
						<enumeratedValue>
							<name>DISABLED</name>
							<description>Disabled. Peripheral DMA requests are disabled.</description>
							<value>0</value>
						</enumeratedValue>
						<enumeratedValue>
							<name>ENABLED</name>
							<description>Enabled. Peripheral DMA requests are enabled.</description>
							<value>1</value>
						</enumeratedValue>
					</enumeratedValues>
				</field>
				<field>
					<name>HWTRIGEN</name>
					<description>Hardware Triggering Enable for this channel.</description>
					<bitRange>[1:1]</bitRange>
					<enumeratedValues>
					<name>ENUM</name>
						<enumeratedValue>
							<name>DISABLED</name>
							<description>Disabled. Hardware triggering is not used.</description>
							<value>0</value>
						</enumeratedValue>
						<enumeratedValue>
							<name>ENABLED</name>
							<description>Enabled. Use hardware triggering.</description>
							<value>1</value>
						</enumeratedValue>
					</enumeratedValues>
				</field>
				<field>
					<name>RESERVED</name>
					<description>Reserved. Read value is undefined, only zero should be written.</description>
					<bitRange>[3:2]</bitRange>

				</field>
				<field>
					<name>TRIGPOL</name>
					<description>Trigger Polarity. Selects the polarity of a hardware trigger for this channel.</description>
					<bitRange>[4:4]</bitRange>
					<enumeratedValues>
					<name>ENUM</name>
						<enumeratedValue>
							<name>ACTIVE_LOW__FALLING</name>
							<description>Active low - falling edge. Hardware trigger is active low or falling edge triggered, based on TRIGTYPE.</description>
							<value>0</value>
						</enumeratedValue>
						<enumeratedValue>
							<name>ACTIVE_HIGH__RISING</name>
							<description>Active high - rising edge. Hardware trigger is active high or rising edge triggered, based on TRIGTYPE.</description>
							<value>1</value>
						</enumeratedValue>
					</enumeratedValues>
[... 82 lines cut ...]
							<value>0</value>
						</enumeratedValue>
						<enumeratedValue>
							<name>ENABLED</name>
							<description>Enabled. Destination burst wrapping is enabled for this DMA channel.</description>
							<value>1</value>
						</enumeratedValue>
					</enumeratedValues>
				</field>
				<field>
					<name>CHPRIORITY</name>
					<description>Priority of this channel when multiple DMA requests are pending. This description reflects a 3-bit priority field giving 8 priority levels. A specific instance of the SDMA might have anywhere from 2 to 16 priority levels (1 to 4 bits for the CH_PRIORITY field). 0x0 = highest priority. 0x7 = lowest priority.</description>
					<bitRange>[18:16]</bitRange>

				</field>
				<field>
					<name>RESERVED</name>
					<description>Reserved. Read value is undefined, only zero should be written.</description>
					<bitRange>[31:19]</bitRange>
					
				</field>
			</fields>
		</register>

there are 4 fields in this register all named RESERVED. They can't be using them for padding because they are all named the exact same thing. Besides an intelligent parser could easily figure out where the reserved bits are without explicitly defined ranges.

porkybrain, 2015-09-20:
After blowing off steam about NXPs design decisions it time to take the pragmatic approach ;) I implemented a free function `remove_reserved` which should solve this issue.

closed 2015-09-20 by porkybrain