cmsis-svd/cmsis-svd #13: No support for dimElementGroups

porkybrain, 2015-09-18:
the parser does not seem to support dimElementGroups as explained here http://www.keil.com/pack/doc/CMSIS/SVD/html/group__dim_element_group__gr.html

porkybrain, 2015-09-20:
We need to add the `    <dim>6</dim> 
    <dimIncrement>4</dimIncrement> 
    <dimIndex>A,B,C,D,E,Z</dimIndex>` values to `SVDRegister` and parse them. 
The actual duplication could be left to higher layers, in my case kvasir specific code or I could implement duplication here https://github.com/porkybrain/cmsis-svd/blob/master/python/cmsis_svd/parser.py#L141. 

What do you guys think is better?

salkinium, 2015-09-20:
I would leave it to the higher levels, since flattening the register tree removes semantical information, that you might still want to use in the higher layers.
However, I would add the duplication code to this repository, but then let the user apply it where necessary.

porkybrain, 2015-09-20:
@salkinium I agree. I would think something like a free function `duplicate_arrays_of_registers` taking either an array of registers or the entire tree of peripherals and retuning the same tree but with all contained arrays expanded. 

I can see the use case where someone walks the tree, arrives at the array of registers, extracts whatever info they want and expands it. 

The other use case would be the user doesn't care about any extra info and just wants to expand everything.

porkybrain, 2015-09-20:
I implemented the free functions mentioned above. This pretty much wraps up support for dimElementGroups so I'm closing this and we can handle any issues in #12

posborne, 2015-09-21:
Leaving opened until changes are actually in tree.

porkybrain, 2015-09-29:
now it really is supported ;)

closed 2015-09-29 by porkybrain