cmsis-svd/cmsis-svd #14: Vendors use enumeratedValues in odd ways

porkybrain, 2015-09-29:
http://www.keil.com/pack/doc/CMSIS/SVD/html/group__svd__xml__enum__gr.html states that "Defines the constant of the bit-field that the name corresponds to.", however some svd files like https://raw.githubusercontent.com/posborne/cmsis-svd/master/data/Freescale/MKL27Z4.svd define enumerated values like this:
```xml
<register>
          <name>PDDR</name>
          <description>Port Data Direction Register</description>
          <addressOffset>0x14</addressOffset>
          <size>32</size>
          <access>read-write</access>
          <resetValue>0</resetValue>
          <resetMask>0xFFFFFFFF</resetMask>
          <fields>
            <field>
              <name>PDD</name>
              <description>Port Data Direction</description>
              <bitOffset>0</bitOffset>
              <bitWidth>32</bitWidth>
              <access>read-write</access>
              <enumeratedValues>
                <enumeratedValue>
                  <name>0</name>
                  <description>Pin is configured as general-purpose input, for the GPIO function.</description>
                  <value>#0</value>
                </enumeratedValue>
                <enumeratedValue>
                  <name>1</name>
                  <description>Pin is configured as general-purpose output, for the GPIO function.</description>
                  <value>#1</value>
                </enumeratedValue>
              </enumeratedValues>
            </field>
          </fields>
        </register>
```
Its a 32 bit wide field with two possible values... I understand what they mean, you pick the right enum value and then shift it, really they should have defined 32 one bit wide fields instead.

Any way even though I understand what they mean my code doesn't and since Kvasir has super static checking it squawks at me that the value I'm wirting into the register is not the same type as the enum. 

Do we agree that this is a misinterpretation of the CMSIS-SVD on the part of freescale or should I be making my script smarter? 

As a work around I'm disabling the generation of enums if they are narrower than the field and have no default, but thats beside the point.

porkybrain, 2015-09-29:
enumeratedValues::enumeratedValue::name is also supposed to be a "String describing the semantics of the value. Can be displayed instead of the value.". Here, and in many many other files, they use `0` or `1` which is not a valid name in C++ (or any other language for that matter) and is certainly not a string description.

posborne, 2015-09-29:
@porkybrain I've run into the same thing.  This is the dirty hack I tried to put together to generate a name based on the description.  It doesn't really work well enough... https://github.com/posborne/zinc/blob/svd-generator-python/support/svd/svdconverter/generators/zinc_ioregs.py#L38

porkybrain, 2015-09-29:
one thing I have noticed is that if most of the enums description is identical the part that is different is usually a good name:
```xml
            <field>
              <name>KEYEN</name>
              <description>Backdoor Key Security Enable</description>
              <bitOffset>6</bitOffset>
              <bitWidth>2</bitWidth>
              <access>read-only</access>
              <enumeratedValues>
                <enumeratedValue>
                  <name>10</name>
                  <description>Backdoor key access enabled</description>
                  <value>#10</value>
                </enumeratedValue>
                <enumeratedValue>
                  <name>11</name>
                  <description>Backdoor key access disabled</description>
                  <value>#11</value>
                </enumeratedValue>
              </enumeratedValues>
            </field>
```

kblomqvist, 2015-09-30:
Remember to check usage element too. It's not used in above examples, so it's read-write by default. But for NRF51, Nordic defines different usages for enumerated values.

porkybrain, 2015-09-30:
Good Point, I don't think we do that right.

porkybrain, 2015-10-02:
I stumbled accross another indication that the vendors are doing this wrong:
http://www.keil.com/pack/doc/CMSIS/SVD/html/svd__outline_pg.html
"Enumerated Values Level: An enumeration maps an unsigned integer constant to a **descriptive identifier** and, optionally, to a description string. Enumerations are used in C to enhance the readability of source code. Similarly, it can be used by debuggers to provide more instructive information to the programmer, avoiding a lookup in the device documentation."

@0xc0710 @salkinium what do you guys think, could the vendors be motivated to provide more **descriptive** identifiers?