cmsis-svd/cmsis-svd #15: Proposition to refactor this project into a collection of CMSIS related tools for Python

kblomqvist, 2015-10-03:
I would propose to start a new project to provide a collection of CMSIS related tools for Python. For example, the project name could be pycmsis and for the beginning it could include:

- CMSIS-SVD file parser
- _"Peripheral header file"_ generator for different languages from vendor CMSIS-SVD files
  - Would use my [yasha](https://github.com/kblomqvist/yasha) code generator. There's already [a template for Rust](https://github.com/kblomqvist/nrf51.rs/blob/master/nrf51.rs.jinja) and [related Jinja2 filters](https://github.com/kblomqvist/nrf51.rs/blob/master/nrf51.jinja-ext#L45).
- CMSIS-SVD vendor files
  - SVD file collection like in this repository, but the files would be within a separate project. So if someone just wants to use the CMSIS-SVD parser and is not interested in to have all the vendor SVD files installed too it would be handy if it can be chosen. Installation of the current project take quite a lot of time just because of all the included SVD files.
  - `require_extras` can be used in `setup.py`.
- CMSIS-DAP
  - Installs [pyOCD](https://github.com/mbedmicro/pyOCD)

Finally, I would like to change the parser. Don't get me wrong. I don't mean to be rude or sound bold, but I think that [the parser I have implemented](https://github.com/kblomqvist/nrf51.rs/blob/master/pycmsis/svd.py) is more elegant and easier to work with.

What do you think?

salkinium, 2015-10-03:
:+1: for using Jinja2 template engine.
Regarding improving the parser, also see the discussion with @porkybrain in the other raised issues (#8, #10, #11, #13, #14).

porkybrain, 2015-10-03:
Our requirements from a kvasir standpoint are pretty much met at this point, provided we make a Version 2.0 release so its easy for other parties to install. 

I will have a look at jnja2, I'm not really familiar with it at this point. 

I'm for anything that gets more people aligned towards the same goal, so yes an organization sounds like a good idea.

posborne, 2015-10-04:
I get a little nervous about the idea of "starting a new project" and think it should only be done if what exists really does not make sense.  I think this project has a well-defined purpose as described in the readme: https://github.com/posborne/cmsis-svd#what-is-this and the current implementation is a very good start towards achieving the stated goals of the project.

On to the concrete items:

> CMSIS-SVD file parser
> ...
> Finally, I would like to change the parser. Don't get me wrong. I don't mean to be rude
> or sound bold, but I think that the parser I have implemented is more elegant and easier
> to work with.

I think there is room for different parser implementations, even within a language ecosystem.  The parser I put together (with contributions from @porkybrain) is admittedly not clever but it is also very easy to reason about and does a decent job of separating concerns.  Arguments related to "elegance" are not generally terribly fruitful (particularly without justification) but I would be interested in ideas for exposing any API that is more usable.  I hesitate to just throw away the API that exists as I know of a few users making use of what already exists.

Starting out, I may change the project description to mention the inclusion of "reference" parser implementation rather than canonical ones.

> "Peripheral header file" generator for different languages from vendor CMSIS-SVD files
> 
> Would use my yasha code generator. There's already a template for Rust
>    and related Jinja2 filters.

This is outside of the stated goals for this project and makes sense as a different project.

> CMSIS-SVD vendor files
> SVD file collection like in this repository, but the files would be within a separate project. So if
>  someone just wants to use the CMSIS-SVD parser and is not interested in [...]

Creating a separate project does not seem necessary.  I previously was on the fence with regard to whether the data files should be included with the Python parser.  They are currently included in order to make things easier for users, but I could just be making up a use case that really doesn't exist.  I would be interested to hear other's thoughts on whether or not the data files should be included with the Python package for the parser.

> CMSIS-DAP
> Installs pyOCD

What would the point of this project be?  I guess I don't get it.  Is this not something that could be handled via contributions to pyOCD?

-------------------------

When creating various projects, I think we could probably create some cohesion and gain some network effects by adding links between them.  I would accept a PR, for instance, updating the README to point to Yasha.

porkybrain, 2015-10-04:
I currently like the fact that all the files are included. Since this repo is now a dependency for any serious kvasir devs and I hate fiddling with dependencies its nice that its easy to install.

porkybrain, 2015-10-05:
After looking at @kblomqvist implementation I second the elegance arguement (full disclosure I'm still a python noob). I'm trying to implement support for the peripheral `derivedFrom` property and it seems like I have to write far more code than is needed with his archetecture.

kblomqvist, 2015-10-05:
"The start of the new project" would be more like the merge of your and my efforts and knowledge. I would really much like to contribute to a common project than working alone. There seems to be a nice community building up here. However, as much as I would like to help you and e.g. add support for size, access and derived peripherals mentioned by @porkybrain, I can't do that without huge refactoring. So that's why I'm proposing to switching the parser. The end-user API is not the problem but the architecture... Again, as bold as it sounds, I think that adding features to "my parser" is much easier.

Regarding to "the peripheral file" creation, why it couldn't be the goal of the project. Like you prefer having the vendor SVD files easily available, why not also provide template files and way to express the parsed SVD files in another file. My idea of having data (vendor SVDs) in separate PyPI project, can be also done in a way that the umbrella project installs that data by default, but there's option to not fetch the files. That way it's also very clear which parts of the projects are licensed by whom.

Next pyOCD. I wasn't aware that there's CMSIS-DAP implemented for Python, until last week. With pyOCD you can run CMSIS-DAP debugger even in Python interpreter. So the idea of making the umbrella project to install it let people using Python aware of other CMSIS related things. Of course the contributions are done directly to pyOCD.

Lastly, if this just doesn't make sense, let's cross reference our efforts at least in README.

closed 2015-12-21 by kblomqvist