cmsis-svd/cmsis-svd #16: How to deal with doubble tags?

porkybrain, 2015-10-05:
In MKL25Z4.swd:
 ´´´xml
  <cpu>
    <name>CM0PLUS</name>
    <revision>r0p0</revision>
    <endian>little</endian>
    <mpuPresent>false</mpuPresent>
    <fpuPresent>false</fpuPresent>
    <mpuPresent>false</mpuPresent>
    <vtorPresent>true</vtorPresent>
    <nvicPrioBits>2</nvicPrioBits>
    <vendorSystickConfig>false</vendorSystickConfig>
  </cpu>
 ´´´

´mpuPresent´ is defined twice, how should we handle stuff like this generally? Here we can ignore it because they are the same, what if they were different?

kblomqvist, 2015-10-05:
I would use the first one and maybe introduce parser output messages. A warning message something like this could be saved within the parser class: `Warning: cpu.mpuPresent defined twice.`

posborne, 2015-10-05:
I believe the current behavior would be to use the first (https://docs.python.org/2/library/xml.etree.elementtree.html#xml.etree.ElementTree.Element.find).  I'm not sure if it is worth it to do the extra work to warn that multiple exist (would increase parse times some).

Might be nice from the perspective of validating SVDs for vendors.  In theory, they could get the same by validating against the XSD (I haven't tried).

kblomqvist, 2015-10-05:
Does the current parse time takes too long or why to consider parse time? I see the sanity checking only as an added complexity, and as you pointed out, most likely it's not worth to do so far.

posborne, 2015-10-05:
Parse time probably isn't too big of a deal.  Running the tests takes awhile (parses every SVD) but you can do it in parallel which makes things move along a bit faster.  Since this stuff is mostly going to be used in code generators and the like, heavily optimizing parsing is probably a premature optimization.  We could probably move to lxml and get some performance enhancements for free.