cmsis-svd/cmsis-svd #17: No support for inheriting tags

porkybrain, 2015-10-05:
Tags like `derivedFrom` or `size` etc. affect children further down the tree. In order to implement support for them we need to either do a second pass through the parsed tree and fill in the default values when possible or make the tree which has been parsed so far availible to the `_parse_x(self, x_node)` node functions to allow them to extract defaults from further up the tree. 

As I understand so far it is more pythonical to do just one pass, should I do it that way?

kblomqvist, 2015-10-05:
I think it's not more "pythonical" to do one pass parsing. It might even be that the `derivedFrom` element is allowed to be defined before it's base element. If that's the case then one pass parsing is not even possible?

`size` and `access` are relatively easy to support if the parent element fields can be passed as defaults for the current one.

Btw. so far I have implemented only derived peripherals. I have not yet found a good solution how to support derived `register`, `field` and `enumaratedValue`. And that it wouldn't be too easy `cluster` can be also derived :)

posborne, 2015-10-05:
One approach would be the following:

1. First pass just parses the raw items and stores them against the appropriate parts of the tree.
2. Methods/Properties do computation on request.  Caching may be utilized in order to speed up cases where calculation is expensive and multiple requests for the same data are expected.

With this approach, the second pass is lazy and only done if required.

porkybrain, 2015-10-07:
@posborne So that would mean adding a `parent` to each object in model.py and adding getters to all properties which could have defaults higher up the tree?

I my use case I want pretty much all the information availible in the tree, so being lazy dosn't really buy me much but I could imagine many other use cases.

posborne, 2015-10-07:
Yeah, with my idea you would have to store a reference to the parent or similar.  In general, I like to reduce the amount of state that has to be stored where possible as I find it leads to fewer problems.  I don't want to restrict the how of the feature implementation too much starting out as I would rather have a less-than-ideal implementation than a perfect one that does not exist.