cmsis-svd/cmsis-svd #19: Inheritance not fully defined

porkybrain, 2015-10-07:
I'm having trouble interpreting effects of the derivedFrom tag. It is not clear to what granularity tags in the derived ovject override tags in the base object. For example here http://www.keil.com/pack/doc/CMSIS/SVD/html/group__svd__xml__peripherals__gr.html the peripheral "Timer0" has one interrupt called "TIM0_INT", "timer1" derived from "timer0" which begs the questions: 
 - Does "timer1" have no interrupts?
 - Does "timer1" share an interrupt with "timer0"? If so why is it called "TIM0_INT"?

porkybrain, 2015-10-07:
What do you guys think about this solution regarding interrupts:
 - `<interrupts>` tag missing completly means inherit everything from the base
 - `<interrupts>` tag exists but has not derived tags means the peripheral has no interrupts
 - '<interrupts>' tag has derived tags means do not inherit any interrupts

the last one is important because there is no other way to distinguish that these tags should not be inherited (which is probably the most common case).

kblomqvist, 2015-10-07:
Good point. The example isn't very good and the interrupt would be shared there, I would say. But practically it would seldom be the case.

porkybrain, 2015-10-07:
I don't see the case where a peripheral inherits an interrupt as being common. Really only in the case where they share. On the other hand there are probably few peripherals which do not have interrupts so it really doesn't matter what we choose it will probably only bite us in corner cases.

posborne, 2016-01-27:
@porkybrain Did you end up making any more sense of this?

porkybrain, 2016-01-27:
Not yet, I have not gotten around to handling interrupts in a generic way in Kvasir so I have had no need so far to parse them correctly. I hope to talk about this with a few people at FOSDEM this coming weekend.

posborne, 2016-03-07:
In the latest PR, I took a guess as to how this maybe should work.  Any news from FOSDEM?