cmsis-svd/cmsis-svd #23: LPC11xx_v6.svd has multiple enumerated values with the same name

porkybrain, 2016-01-08:
So far we have been correcting names of things in extention files in order to not mess with the SVD (in order to stay open to drop in improvements from the vender), however since the extention files index things by name ambiguities are impossible to correct. 

The offending tag is ```<name>COUNTER_MODE_TC_IS_</name>``` which occurs three times in the CTM field of the CTCR register all with different values.

What should we do to fix this?

kblomqvist, 2016-01-08:
Just an idea... maybe you could append the value into the name. Either in a separate post-process step or in all cases where the last character of the name is "_".

COUNTER_MODE_TC_IS_1
COUNTER_MODE_TC_IS_2
COUNTER_MODE_TC_IS_3

porkybrain, 2016-01-08:
good idea, maybe we could also make an alternative indexing mode in the extension files too. I'm a little reluctant to make too many changes because of this because in my mind it is simply an NXP error. I'm also considering just changing the svd file in this case.

Samonitari, 2017-09-07:
Hi all!
I know this is an old issue/thread, but seemingly nobody is tending to it.
Most  NXP LPC SVD's are obviously autogenerated from the docs, with truncating, so this issue is present in them. Either the header generator won't eat them, or their output is unchecked, and won't work.
I made corrections in LPC82X.svd - which is not in this repo by the way - and will definitely do the same for some others.

So if no objections, I would like to hijack this issue, and document here the progression.

odinthenerd, 2017-09-07:
go for it, glad to here someone is working on it!