cmsis-svd/cmsis-svd #27: is this the place to maintain SVD files?

hackrid, 2016-04-20:
Hi,

I'm working with embsysregview and have edited some svd files.
is this the most popular place to "upstream" such changes?

best reagards

0xc0170, 2016-04-20:
> I'm working with embsysregview and have edited some svd files.

Can you elaborate? What have you changed? embsysregview contains those svd files.

>  is this the most popular place to "upstream" such changes?

Did you consider talking directly to vendors of those ``svd`` files? They could update their cmsis packs.

hackrid, 2016-04-20:
I have fixed mainly descriptions, read/write issues and missing core registers.

i have little to no hope at all to get a response from the vendors in  a timely manner. I have contacted ST before and did not get a response in months. 

because of this, I would find it really nice if there was a public repo containing the most up to date files with corrections from people wanting to be correct :)

posborne, 2016-04-20:
Hi @hackrid.  Currently, we are focused on geting updates directly from Vendors so things don't get too out of sync, but we could revisit this if Vendors are behaving too poorly.

For now, I am optimistic that the combination of bullying from @0xc0170 and others at ARM and a little public shaming could be useful.  I would welcome a PR against this repo with the update and hopefully we can find somebody within ST to take a look and get things updated within their CMSIS pack in sync.

I think that for many silicon vendors, the SVD files are probably not an an artifact that they edit directly as IP chunks are reused across many processors.

niklas-arm, 2016-04-20:
@ReinhardKeil and his team maintain everything related to [CMSIS](https://github.com/ARM-software/CMSIS) and is probably the best person to ask about SVD related updates and fixes for vendors.

hackrid, 2016-04-21:
thanks for your responses!
having some people at ARM / vendor site to support the efforts of improved svd files would surely help!

@posborne which way of public shaming do you consider most useful? before I issue a PR, I should spread the added information across the whole ST devices. But this is some effort I can't afford today.

aethaniel, 2016-04-29:
Hi, Atmel delivers SVD files as part of DFP here: http://packs.download.atmel.com/.
AFAIK, any issue can be reported using the Atmel support available via Atmel web.