cmsis-svd/cmsis-svd #34: Installation failure with some versions of pip

c1728p9, 2016-08-10:
I'm seeing an installation failure when installing cmsis-svd version 0.2 or 0.3. Version 0.1 installs without problems. This can be reproduced with the following script:
```
virtualenv env
env\Scripts\activate
pip install pip==7.1.2
pip install cmsis-svd==0.3 --no-cache
```

Related issue https://github.com/mbedmicro/pyOCD/issues/251

c1728p9, 2016-08-11:
Hi @posborne, I can confirm #35 fixes the problem I was seeing. Can you push a new release to pypi so I can update the dependencies in pyOCD?

posborne, 2016-08-11:
Version 0.4 is released.  Looks like something didn't quite work right with conversion to rst on the readme, but whatever.  Not a big deal.  I can fix that later.

https://pypi.python.org/pypi/cmsis-svd/0.4

closed 2016-08-11 by posborne