cmsis-svd/cmsis-svd #73: Push a new version to https://pypi.org/project/cmsis-svd/

bradjc, 2019-06-04:
The last release on PyPI was in 2016.

bradjc, 2019-11-07:
Given the many recently merged PRs, a new release would be helpful for making this easy to use.

ppannuto, 2020-04-09:
@posborne Is there any timeline or blockers we can help with for pushing a new version to PyPI?

The Tock project is considering forking and pushing our own version to PyPI as we need some of the updates, but we would prefer to just use the official version if possible.

posborne, 2020-04-09:
Hi @ppannuto, we can move something forward.  One items that has been discussed has been removing the SVD from the pypi package and having it just be the python code and relying on direct Github dep for the XML.  Would this work with your usage?

ppannuto, 2020-04-15:
Umm, I think that would be fine. It's largely orthogonal to our concerns I think. 

We're mostly interested in having a released version that includes this change: https://github.com/posborne/cmsis-svd/pull/56

A release that pushes the current state of the repository to PyPI now, deferring the SVD data question, would be much more valuable.

kyberdin, 2021-10-04:
Has there been any further effort on this? A modern release that includes support for #56 would be very helpful.

pengi, 2023-05-09:
Hi,

Would really like this patch too. I'm using this project in my arm-gdb tool, and the cluster support would be really helpful, since a lot of registers won't show up in many platforms. Cluster support is for what I can see something that was merged 5 years ago, back in 2018

Great tool though, really like it.

VincentDary, 2024-10-25:
Work In Progress, see [#192](https://github.com/cmsis-svd/cmsis-svd/pull/192)