cmsis-svd/cmsis-svd #157: Is this repo abandoned?

michalfita, 2022-07-13:
It's been four months since last commit to this repository and the queue of PRs is now quite long. People decided to rely on this as source of truth regarding SVDs, but in the context of that delay it seems it was bad decision.

Any chances to get these picked up soon? If not maybe consider transfer responsibility into some other hands, I suggest create organisation with a few seriously interested parties as maintainers.

Thank you.

posborne, 2022-07-14:
@michalfita Not abandoned, I was on an extended backpacking trip for awhile and haven't fully caught up on things.  I'm very open to adding additional maintainers to the project and potentially moving to another org.  I'm also involved in the rust-embedded org as a maintainer there but feel this project probably should remain separate in some form given its more broad appeal.

I'll see about working through some of the backlog today; some of the older PRs and such are on hold due to lack of clarity around changes to files not provided by vendors and such that we have generally tried to avoid, etc. but I'm definitely behind some on the more recent ones.

michalfita, 2022-07-15:
Thanks, @posborne, for the update.

What I did with my PAC/HAL crate repos is I moved it to the organisation and invited parties most interested in doing the job. I assume any senior programmer would responsibly take role in there for the sake of themselves and rest of the community. You could create something like `svd-catalogue` organization and invite most active participants as maintainers. The organization can hold repositories for other tools related to SVD and even fork other repos to participate in PRs.

closed 2022-07-15 by michalfita