cmsis-svd/cmsis-svd #158: parsing blob files, possible with this lib?

hanetzer, 2022-09-04:
Hello. Dealing with some hisilicon chips. they generally initalize a number of registers using a sort of binary
csv file with the following format:
```c
struct regentry {
    uint32_t reg_addr; // address of the register
    uint32_t value; // value to be written or read
    uint32_t delay; // amount of nop's to execute to delay
    uint32_t attr; // read or write and other flags
```
I've been putting together a svd file for the soc for use with the PyCortexMDebug project, and was hoping
this may allow for me to use said svd file to parse out the actions taken instead of manually figuring the bitfields.

for reference:
```hex
00000000  80 00 01 12 00 00 00 00  00 00 00 00 fd 00 00 00  |................|
00000010  e0 01 01 12 00 00 00 00  00 00 00 00 00 00 55 00  |..............U.|
00000020  00 00 00 00 00 00 00 00  64 00 00 00 00 00 00 00  |........d.......|
00000030  10 00 01 12 55 55 d5 12  00 00 00 00 fd 00 00 00  |....UU..........|
00000040  14 00 01 12 2d 10 00 00  00 00 00 00 fd 00 00 00  |....-...........|
00000050  00 00 00 00 00 00 00 00  64 00 00 00 00 00 00 00  |........d.......|
00000060  e8 01 01 12 0f 00 00 00  00 00 00 00 00 00 1d 00  |................|
00000070  80 00 01 12 9b 05 00 00  00 00 00 00 fd 00 00 00  |................|
00000080  e0 01 01 12 9b 05 00 00  00 00 00 00 00 00 55 00  |..............U.|
00000090  00 00 00 00 00 00 00 00  64 00 00 00 00 00 00 00  |........d.......|
```

Note these values are encoded in little endian.

BenBE, 2022-09-04:
This is possible. Had a project,where I got a list of addresses for my MCU where I needed to translate those addresses back to a readable name.

The general outline is to first iterate over all peripherals, collect their memory range and if your address lies within that range, iterate over all of the peripheral's registers (by adding their offset to the peripheral's base). If you find a match there, than your address belongs to the peripheral and relates to the register you found.

There are ways to improve this general algorithm by building a search tree over those values you have to check, but the general idea stays the same.

hanetzer, 2022-09-04:
Thanks @BenBE. I'm not much of a pythonist, do you think you could provide a small snippet as
an example of how one would do this?

hanetzer, 2022-09-07:
Awright. I've gotten to the point where I have a reg address (Peripheral base address + register address) and
a SVDRegister object, and value read/written, along with bitrange, in a context. Now I just need to figure out
how to parse out the bitrange of the value into fields and their enumerated values, if possible. hrm.