cmsis-svd/cmsis-svd #164: Please make a new release in PyPI.

markrages, 2022-12-22:
Over the last six years, many new SVD files have been added that are not included in the currently available release.

posborne, 2023-01-09:
Given the size to which things have grown, I am leaning toward doing a new pypi release of a major version but which would not include the SVD definitions themselves in the pypi artifact.  Though this is less convenient, I think it is more appropriate in general.

darryllawson, 2023-01-31:
Firstly, thank you to all the contributors of this excellent package!

We would like to have a new version (specifically with register cluster support) available on PyPI too.

I agree that excluding the SVD definitions from the PyPI package probably makes sense (we do not make use of those anyway).

markrages, 2024-01-02:
Version 0.0.6 of https://github.com/markrages/svd_gdb now downloads the SVD files on demand, so this is no longer relevant.

closed 2024-01-02 by markrages