cmsis-svd/cmsis-svd #169: SVD data model inconsistency due to Python attribute dynamic lookup for SVD "drivedFrom" tags

VincentDary, 2023-02-18:
Hello,

I pinpoint here the inconsistency I have found with the actual SVD data model implementation and the Python object attribute dynamic lookup for derivedFrom SVD tag. 

For example, with the following SVD file `cmsis-svd/data/Toshiba/M367.svd`, UART5 is derived from UART4.

```xml
<peripheral derivedFrom="UART4">
  <name>UART5</name>
  <baseAddress>0x40049000</baseAddress>
</peripheral>
```

In the actual data model implementation if we look for the parent peripheral base address of the data register of UART5 (UART5:DR) the value is `0x40048000` because the implementation use `_lookup_possibly_derived_attribute` for data retrieving, see the example below. However, this is semantically wrong for at least two reasons: firstly, the base address of UART5 parent of the register UART5:DR is `0x40049000`, secondly, the parent object of SVDRegister UART5:DR is the SVDPeripheral object corresponding to UART5, not the SVDPeripheral object corresponding to UART4.

```text
>> from cmsis_svd.parser import SVDParser
>> parser = SVDParser.for_packaged_svd('Toshiba', 'M367.svd')
>> parser.get_device().peripherals[6].name
'UART5'
>> parser.get_device().peripherals[6].registers[0].parent.name
'UART4'
>> hex(parser.get_device().peripherals[6].registers[0].parent.base_address)
'0x40048000'
```

I think this is a problem because it is not possible to build a flat register list and retrieve the correct address of a register in this list when It comes from a derived peripheral. For example, the semantic of the following code is right but the resulting output is wrong.

```text
>>> from cmsis_svd.parser import SVDParser
>>> parser = SVDParser.for_packaged_svd('Toshiba', 'M367.svd')
>>> registers = []
>>> for p in parser.get_device().peripherals:
...     registers.extend([r for r in p.registers])
...
>>> for r in filter(lambda x: x.name == 'DR', registers):
...     if r.name == 'DR':
...             print(f'{r.parent.name} {r.name} '
...                   f'{hex(r.parent.base_address + r.address_offset)}')
... 
SSP0 DR 0x40040008
SSP0 DR 0x40040008
SSP0 DR 0x40040008
UART4 DR 0x40048000
UART4 DR 0x40048000
```

In the simplest case, I think it would be better to copy the whole source object for a derived object and update only the changes in the derived object. The memory footprint will be a bit larger but the semantics will be preserved. I can propose to write a patch that goes in this direction, however, this implies many changes, and especially to remove the dynamic attribute resolution. Are you agree with all these changes ? If yes I can submit a patch...

See you soon.

pengi, 2023-05-09:
I got an issue related to this too. In my case, it seems that register clusters isn't working on the derived set. Haven't tracked down the problem exactly yet, which I plan to do. But it seems that for nrf52840, the `SPIS0` peripheral has registers in the `RXD` and `TXD` register cluster, but the derived peripherals `SPIS1` and `SPIS2` doesn't have them.

I personally agree with keeping it simple and consistent is worth more than the memory footprint, if it reduces the amount of bugs. So I personally like the idea of copying instead of keeping up at runtime in the case of `derivedFrom`

The issue can be seen using the tool `arm-gdb` (from pypi), when using cmsis-svd from git:

```
(gdb) arm list nrf52840 SPIS0
Registers in SPIS0 @ 0x40003000:
 - TASKS_ACQUIRE @ +0x24
        ...............................# TASKS_ACQUIRE
 - TASKS_RELEASE @ +0x28
        ...............................# TASKS_RELEASE
[...]
 - RXD_PTR @ +0x534
        ################################ PTR
 - RXD_MAXCNT @ +0x538
        ................################ MAXCNT
 - RXD_AMOUNT @ +0x53c
        ................################ AMOUNT
 - RXD_LIST @ +0x540
        ..............................## LIST
 - TXD_PTR @ +0x544
        ################################ PTR
 - TXD_MAXCNT @ +0x548
        ................################ MAXCNT
[...]
```
but for `SPIS2`, this is all:
```
(gdb) arm list nrf52840 SPIS2
Registers in SPIS2 @ 0x40023000:
 - TASKS_ACQUIRE @ +0x24
        ...............................# TASKS_ACQUIRE
 - TASKS_RELEASE @ +0x28
        ...............................# TASKS_RELEASE
 - EVENTS_END @ +0x104
        ...............................# EVENTS_END
 - EVENTS_ENDRX @ +0x110
        ...............................# EVENTS_ENDRX
 - EVENTS_ACQUIRED @ +0x128
        ...............................# EVENTS_ACQUIRED
 - SHORTS @ +0x200
        .............................#.. END_ACQUIRE
 - INTENSET @ +0x304
        ..............................#. END
        ...........................#.... ENDRX
        .....................#.......... ACQUIRED
 - INTENCLR @ +0x308
        ..............................#. END
        ...........................#.... ENDRX
        .....................#.......... ACQUIRED
 - SEMSTAT @ +0x400
        ..............................## SEMSTAT
 - STATUS @ +0x440
        ...............................# OVERREAD
        ..............................#. OVERFLOW
 - ENABLE @ +0x500
        ............................#### ENABLE
 - CONFIG @ +0x554
        ...............................# ORDER
        ..............................#. CPHA
        .............................#.. CPOL
 - DEF @ +0x55c
        ........................######## DEF
 - ORC @ +0x5c0
        ........................######## ORC
```

This tool reads the `peripheral.registers` parameter, which should resolve clusters, but apparently not.

pengi, 2023-05-09:
Oh, my issue (possibly not the original in the thread) might have been fixed in PR #100 though. Even though my thought on keeping the structure simple still applies :)

closed 2023-10-09 by VincentDary