cmsis-svd/cmsis-svd #173: No iterable field element return, if a fields name contains the string 'reserved'

BrinkmannJan, 2023-06-16:
## Expected Behavior
The parser returns an iterable object, regardless of the name of the bitfield.

## Actual Behavior
If there's the string 'reserved' anywhere in the name of a field the parser will not return an iterable element.

## SVD example of the the Problem
The attached archive contains a .svd file and a script generated C-Header.

[reserved_example.zip](https://github.com/posborne/cmsis-svd/files/11771628/reserved_example.zip)

## Context
This described behavior makes it very difficult for me to write my own script (When there's know iterable object how should I detect that there should have been one?).
Is there a specific default in the svd file format that specifies the use of the 'reserved' tag and explains this behavior.
If I use the SVDConv.exe from https://github.com/Open-CMSIS-Pack/devtools/tree/main/tools/svdconv/SVDConv my example svd file will be parsed correctly.

ckudera, 2023-07-17:
@BrinkmannJan I tried to reproduce your issue, but for me the parsing works as expected although I used you provided example. Could you please provide a code sample?

BrinkmannJan, 2023-09-12:
@ckudera Thanks for your response! I found out that using the latest dev version of the parser (instead of 0.4) solves this problem for bitfields containing the string "reserved".

In version 0.4 the following line of code was missing the **not** ("if **not** self.remove_reserved ...").
https://github.com/posborne/cmsis-svd/blob/0dd8f9af9a69399088a711259af813383bef08d3/python/cmsis_svd/parser.py#L263

This Bugfix solves my described issue.  Of course as long as you do not mess with "remove_reserved" which is initialized with "false".

closed 2023-09-12 by BrinkmannJan