cmsis-svd/cmsis-svd #188: XMC7200 svd parsing problems

rtmagj, 2024-02-06:
[cat1c8m.zip](https://github.com/cmsis-svd/cmsis-svd/files/14176603/cat1c8m.zip)

Trying to parse attached svd with development version of cmsis-svd but ends with 
```
  File "d:\test.py", line 59, in svd
    for peripheral in parser.get_device().peripherals:
                      ^^^^^^^^^^^^^^^^^^^
  File "C:\Users\abcd\src\cmsis-svd\python\cmsis_svd\parser.py", line 733, in get_device
    return self._parse_device(self._root)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\abcd\src\cmsis-svd\python\cmsis_svd\parser.py", line 692, in _parse_device
    peripherals.append(self._parse_peripheral(peripheral_node))
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\abcd\src\cmsis-svd\python\cmsis_svd\parser.py", line 629, in _parse_peripheral
    return SVDPeripheral(
           ^^^^^^^^^^^^^^
  File "C:\Users\abcd\src\cmsis-svd\python\cmsis_svd\model.py", line 384, in __init__
    for r in _none_as_empty(self.registers):
                            ^^^^^^^^^^^^^^
  File "C:\Users\abcd\src\cmsis-svd\python\cmsis_svd\model.py", line 399, in registers
    regs.extend(cluster.registers)
  File "C:\Users\abcd\src\cmsis-svd\python\cmsis_svd\model.py", line 330, in registers
    yield self.updated_register(reg, cluster, i)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\abcd\src\cmsis-svd\python\cmsis_svd\model.py", line 305, in updated_register
    name="{}_{}".format(clu.name % i, reg.name),
                        ~~~~~~~~~^~~
TypeError: not all arguments converted during string formatting
```

After removing CANFD peripherals there still seems to be a problem in PERI peripheral with strings that have dual pattern replacements.

VincentDary, 2024-09-30:
Thank you very much for your feedback. But, for safety reasons about the potential bug exploitation in decompression tools, I accept only plain text file. Can you provide your SVD file in plain text, and host this file on github.com, please ?

BenBE, 2024-09-30:
Have fun …
[cat1c8m.svd.txt](https://github.com/user-attachments/files/17193797/cat1c8m.svd.txt)

VincentDary, 2024-09-30:
Thank's @BenBE