cmsis-svd/cmsis-svd #191: Provide SVD packs as packaged features

BenBE, 2024-10-07:
It would be nice, if one could pre-install some SVDs bundles as part of the package installation with PIP such that you could install `cmsis-svd[stm]` if you regularly use MCUs by ST Micro and thus have those SVDs available locally. If a given MCU cannot be found locally, there could/should still be a fallback to the online lookup provider as it is now.

These features could for example be done per vendor or per major device series (to stick with the example of STM: have packages for stm32f0, stm32f1, stm32l0, …) with "vendor features" being an alias depending on all of the vendor's series.

This feature was first proposed to @VincentDary OTR and copied here for further, regular tracking of this idea. Comments and further refinements welcome.

VincentDary, 2024-10-14:
Hello @BenBE,

I think this is very important to add this feature quickly, since cmsis-svd brought a high value to developer by providing built in cmsis-svd files.

I propose to add this feature at install time by using the [`cmdclass`](https://setuptools.pypa.io/en/latest/deprecated/distutils/apiref.html#distutils.core.setup) argument of `setuptools.setup` in `setup.py`, and implement a custom **download command** which subclasses [distutils.core.Command](https://setuptools.pypa.io/en/latest/deprecated/distutils/apiref.html#distutils.core.Command).

A good exemple exist in the `setup.py` of binwalk for the `idainstall` option:
- import: 
   https://github.com/ReFirmLabs/binwalk/blob/cddfede795971045d99422bd7a9676c8803ec5ee/setup.py#L348
- cmdclass:
   https://github.com/ReFirmLabs/binwalk/blob/cddfede795971045d99422bd7a9676c8803ec5ee/setup.py#L351C23-L351C40
- command implementation:
   https://github.com/ReFirmLabs/binwalk/blob/cddfede795971045d99422bd7a9676c8803ec5ee/setup.py#L133
- usage: 
   https://github.com/ReFirmLabs/binwalk/wiki/Using-the-Binwalk-IDA-Plugin#installation

For exemple:

```python

from setuptools import Command, setup

class CmsisSvdBundleInstall(Command):
    description = "Installs CMSIS-SVD bundle."
    user_options = [
        ('bundle=', None, 'Specify a list of bundle to install.'),
    ]

    def initialize_options(self):
        self.bundle = None

    def finalize_options(self):
        pass

    def run(self):
        bundle_to_install = self.bundle.split(',')
        # implement me

setup(
    cmdclass={
        'bundleinstall': CmsisSvdBundleInstall
    }
    ...
)

```

usage:

```text
python setup.py bundleinstall --bundle=stm,Atmel
```

I don't really know how to transfer these arguments via `pip install` but normally It won't be a problem.

Refinements are welcome :) !

BenBE, 2024-10-14:
Would having separate packages `cmsis-svd-data-atmel`, `cmsis-svd-data-stm`, … feasible?

This would allow `pip install cmsis-svd[stm]`, which would then also pull `cmsis-svd-data-stm`. But that would necessitate having several additional packages being created (@brainstorm).

VincentDary, 2024-10-14:
IMHO, I think make a package by constructor is to much work.

Maybe we can generate an index.json at the root of cmsis-svd-data repository, which reference all svd files with a hash , and use a doted notation for the installer to download a specific list of SVD or vendor following the directory tree of cmis-svd-data. 

Exemple to download a specific SVD file:
- `STMicro.STM32F20x` [cmsis-svd-data/data/STMicro/STM32F20x.svd](https://github.com/cmsis-svd/cmsis-svd-data/blob/main/data/STMicro/STM32F20x.svd)
- `Freescale.MK10D10` [cmsis-svd-data/data/Freescale/MK10D10.svd](https://github.com/cmsis-svd/cmsis-svd-data/blob/main/data/Freescale/MK10D10.svd)

Exemple to download multiples specifics SVD files:

- `STMicro.STM32F20x,Freescale.MK10D10`

Exemple to download a specific file and all Freescale SVD:
- `STMicro.STM32F20x,Freescale`

Also, this is possible to develop a downloader python command script which allow to download the last SVD update according the hash store in the index.json. By this way we are not dependant of the install step to download/update SVD files.  Also the index.json can be used to list available constructor and available SVD file.

VincentDary, 2024-10-15:
@BenBE , @brainstorm , Below a little POC for the downloader, give me your feedback if your are interested or not, it's quite slow, but I think that my download routine is bad, or maybe Github restricted the bandwidth.

[svd-dl poc](https://github.com/cmsis-svd/cmsis-svd/blob/svd-downloader/python/cmsis_svd/svd_dl.py)

usage to download everything:

```text 
svd-dl download --svd-resources=ALL
```

Usage to dowload specific svd ressources:
```text 
svd-dl download --svd-resources=STMicro.STM32F20x,Freescale
```

BenBE, 2024-10-16:
I'll look into this a bit closer later on, but from a quick glance I noticed two potential issues:

1. Downloading many files: Transferring a file with HTTP requires both a TLS handshake and the so-called TCP slow start. Depending on latency these can incur quite a delay and slowdown.

2. Each HTTP request has a certain latency over the network.

Thus you want to re-use one (or few) connection(s) and reduce the number of HTTP requests performed.

I'll later give it a try and see what I can manage.

BenBE, 2024-10-16:
Okay, just with using `.tar.gz` for e.g. SiliconLabs we get down from >4GiB to ~180MiB on the network, which should speed up download significantly. For Atmel I get 405MiB vs. 19MiB, for FreeScale 429MiB vs. 17MiB, and for STMicro we have 196MiB vs. ~10MiB. Thus basically a saving of 20x.

Results with ZStd look even better:
Atmel: 405MiB -> 1.47MiB
FreeScale: 429MiB -> 2.17MiB
NXP: 100MiB -> 1.41MiB
SiliconLabs: 4.07GiB -> 2.34MiB (!!!)
STMicro: 196MiB -> 1.6MiB

With ZStandard this is usually more on the line of ~200x getting better the larger the archives become.

Test source (bash):

```bash
set -xeu -o pipefail

find data/ -type f -name '*.tar*' -print -delete

for a in `seq 1 3` do
   find data/ -mindepth $a -maxdepth $a -type d -exec tar -cf {}.tar.gz -I "zstd --ultra -20" {} \;
   find data/ -mindepth $a -maxdepth $a -type d -exec tar -cf {}.tar.gz -z {} \;
done

true >index.hash;
for a in $(find data -name '*.tar*' -or -name '*.svd'); do
    sha512sum $a >>index.hash
done
```

brainstorm, 2024-10-16:
> Would having separate packages `cmsis-svd-data-atmel`, `cmsis-svd-data-stm`, … feasible?
> 
> This would allow `pip install cmsis-svd[stm]`, which would then also pull `cmsis-svd-data-stm`. But that would necessitate having several additional packages being created (@brainstorm).

I'm late to this party... I'm glad that this went through a custom standalone tool instead of the package management route because it could have complicated things significantly.

Please remember that third parties like the entire Rust Embedded community make use of cmsis-svd too: https://github.com/rust-embedded/svd, so introducing changes to the data files/hierarchy must be considered really carefully. I know changes of this nature have not happened yet, but worth keeping it in mind before it's too late, just in case ;)

BenBE, 2024-10-16:
Based on the `svd-index` branch I updated that script to implement a somewhat more comprehensive index, which contains:

* List of files (using dotted notation)
* List of packages (with content list for each one)

The indexer produces a version of each downloadable file compressed with gzip and one using zstd. Hashes are all using SHA2-512. URLs within the index are relative to a base directory. Files are placed along-side the existing '.svd' files as `.svd.gz` and `.svd.zstd` respectively. Packages are placed in the data root as `packagename.tar.gz` and `packagename.tar.zstd` respectively.

Compressing and indexing the whole archive takes about 10 minutes (CPU time) on my machine. The produced index.json is ~1.6MiB (gzip: 229KiB, zstd: 195KiB) and the additional index files (files+packages) take up abut 1GiB total.

The produced layout is suitable for just referencing the plain files, but an intelligent downloader would try to grab the zstd or gzip versions of each file first or even completely prefer the package archives.

@cmsis-svd/core-maintainers : Can we opt to do feature dev on individual/personal forks/clones of the main repo and only merge to the central repo once done? Currently I see some PRs sourcing from the target repo, which I find somewhat unfortunate, especially as some still are clear experiments.

Find my indexer implementation at https://github.com/BenBE/cmsis-svd-data/tree/svd-indexer. The accompanying downloader based on the PoC will follow later as I get to it.

brainstorm, 2024-10-16:
> Can we opt to do feature dev on individual/personal forks/clones

Why?

BenBE, 2024-10-16:
> > Can we opt to do feature dev on individual/personal forks/clones
> 
> Why?

Mostly two reasons:

1. The main repository is basically the one everyone uses to fork from and monitor upstream changes. In this situation, everything that is published to this repo should not be edited later on. This makes some workflows like rebasing PRs somewhat awkward and conflicts with a separation of workspaces. Git is "distributed" and we should use this decentralized nature to have the "stable" things for everyone and those interested in following the really "hot" stuff can add the remotes of the maintainers (that's also the way things are handled over at htop).

2. It helps to separate which changes can be considered stable (unlikely to be thrown out again) from things like experiments in their early stages, that will likely go through several revisions and refinements before they are integrated. Being able to refine your PRs before merging them tends to give a better/cleaner commit history over all where changes can reflect what needs changing for a feature instead of which labyrinth did you go through while implementing it. The latter can be telling quite a lot too, but having a clean and concise history explaining necessary changes is preferable IMO.

I'm fine either way (will still use my fork though for handing in PRs), but wanted to bring this up, as I think this is something we should at least have discussed.

brainstorm, 2024-10-16:
Good points, thanks! I'll do so on my future PRs ;)

BenBE, 2024-10-19:
Okay, finally came around to hacking some stuff together:

* https://github.com/BenBE/cmsis-svd-data/blob/svd-indexer/indexer.py – Script to generate metadata for providing the SVD resource packs. The contents of the `data/` directory can be served statically as a SVD repository.

* https://github.com/BenBE/cmsis-svd/blob/svd-downloader/python/cmsis_svd/svd_dl.py – Script to download individual files and whole packs of SVD files. The script can take a `--svd-repo=URL` argument to specify alternate SVD repository locations.

Both scripts can optionally support ZStandard for compression when the python environment they are running in provides `pyzstd`. If this package is not available, these scripts silently fall back to use GZip or for individual files uncompressed downloads.

Please review these two scripts and provide feedback.

BenBE, 2024-10-21:
@brainstorm @VincentDary Any feedback on these two implementations?

brainstorm, 2024-10-22:
> @brainstorm @VincentDary Any feedback on these two implementations?

Looks great overall, I'm not too sure about the `.gzip` fallback: I'd stick with just zstd to reduce complexity? Zstd should be supported everywhere in 2024 I reckon.

Also while reading the indexer I was thinking if it'd make sense to run it as a CI job instead of locally and then pushing it manually (which I guess that's how this currently works)?... but then I looked at your SiliconLabs 4GB payload comment and realised that we might not have enough in the GHA free tier to have it automated properly without running into limits quickly :/

trufae, 2024-10-22:
What about xz?

BenBE, 2024-10-22:
> > @brainstorm @VincentDary Any feedback on these two implementations?
> 
> Looks great overall, I'm not too sure about the `.gzip` fallback: I'd stick with just zstd to reduce complexity? Zstd should be supported everywhere in 2024 I reckon.

I explicitly used some sort of fallback (gzip in this case) because ZStandard requires an external dependency for both compression and decompression (`pyzstd`). And actually having multiple compression/decompression formats supported looks worse than it actually is because of the limited structural refactoring I did for this first implementation. If you refactor this to have one class per format and order these accordingly, it quickly boils down to just iterating these classes until the first one succeeds.

> Also while reading the indexer I was thinking if it'd make sense to run it as a CI job instead of locally and then pushing it manually (which I guess that's how this currently works)?...
My implementation is actually intended to have the generated `data/` directory hosted statically somewhere (that's also how I tested things). I didn't post the URL of my repo publicly, because you even could just `python -m http.server data/` and you'd be done …

> but then I looked at your SiliconLabs 4GB payload comment and realised that we might not have enough in the GHA free tier to have it automated properly without running into limits quickly :/

The full repository directory is about 6.5 GiB of required storage to host everything (including all packs and SVDs; both gzip and zstd).

> What about xz?

Implementing other formats shouldn't be too hard, once you do a little refactoring and refinement. Based on [the Python documentation for TarFile](https://docs.python.org/3/library/tarfile.html), support for LZMA has been added in Python 3.3 **if such support has been compiled in**.

I don't see a reason for not supporting XZ given it's usually good compression (better than gzip, comparable to ZStd).

Usually compression quality can be ordered as Gzip < BZip2 < LZMA/XZ < ZStandard.

To summarize:
- XZ is usually built-in to Python 3.3+
- Support for further formats is possible
- To add further formats you would first want to refactor this draft implementation some more
- Regardless we'd probably need some small VPS to place the generated static files (~20 GiB)

VincentDary, 2024-10-22:
> @brainstorm @VincentDary Any feedback on these two implementations?

Of course :) 

Thank you very much for your great proposal an this big POC improvement, below my code review for `indexer.py`:

- Maybe add a `dependency.txt` file at the root of the repository with a `pyzstd` library reference and a frozen version.

- line [202](https://github.com/BenBE/cmsis-svd-data/blob/svd-indexer/indexer.py#L202), [204](https://github.com/BenBE/cmsis-svd-data/blob/svd-indexer/indexer.py#L204): There is a quoting issues which crash the script.

```diff
diff --git a/indexer.py b/indexer.py
index 2cb08a9..14abcdc 100644
--- a/indexer.py
+++ b/indexer.py
@@ -199,9 +199,9 @@ def gen_index_json():
     print(' hash', end='')
     with open(os.path.join(DATA_DIR, 'index.hash'), 'w') as f:
         f.write(f'{os.path.basename(index_filename)} {index_file_plain}\n')
-        f.write(f'{os.path.basename(index_filename) + '.gz'} {index_file_gzip}\n')
+        f.write(f'{os.path.basename(index_filename) + ".gz"} {index_file_gzip}\n')
         if have_zstd:
-            f.write(f'{os.path.basename(index_filename) + '.zstd'} {index_file_zstd}\n')
+            f.write(f'{os.path.basename(index_filename) + ".zstd"} {index_file_zstd}\n')

     print(' done')
```

- line [53](https://github.com/BenBE/cmsis-svd-data/blob/svd-indexer/indexer.py#L53): Code factoring proposal:

```diff
diff --git a/indexer.py b/indexer.py
index 2cb08a9..bc3ac15 100644
--- a/indexer.py
+++ b/indexer.py
@@ -50,10 +50,9 @@ DATA_DIR = os.path.join(os.path.dirname(__file__), 'data')

 def compress_archive_gz(outfile: str, filenames: Dict[str, str]) -> None:
-    with gzip.open(outfile, 'wb', compresslevel=9) as f_out:
-        with tarfile.open(fileobj=f_out, mode='w') as tar:
-            for arcname, source in sorted(filenames.items()):
-                tar.add(source, arcname=arcname)
+    with tarfile.open(outfile, mode='w:gz') as tar:
+        for arcname, source in sorted(filenames.items()):
+            tar.add(source, arcname=arcname)

 def compress_archive_zstd(outfile: str, filenames: Dict[str, str]) -> None:
     with pyzstd.ZstdFile(outfile, 'wb', level_or_option=zstd_options) as compressor:
```

- line [105](https://github.com/BenBE/cmsis-svd-data/blob/svd-indexer/indexer.py#L105): Code factoring proposal:

```diff
diff --git a/indexer.py b/indexer.py
index 2cb08a9..bc3ac15 100644
--- a/indexer.py
+++ b/indexer.py
@@ -102,9 +101,7 @@ def gen_index_json():

         package_for = containing_packages(svd_dot)
         for pkg in package_for:
-            if pkg not in svd_archives:
-                svd_archives[pkg] = {}
-            svd_archives[pkg][svd_file] = svd
+            svd_archives.setdefault(pkg, {})[svd_file] = svd

         svd_size = os.stat(svd).st_size
```

- line [101](https://github.com/BenBE/cmsis-svd-data/blob/svd-indexer/indexer.py#L101): There is a bug at this line because some SVD filename are using dotted version number :/, see below, so IMO unexpected pack are created, I propose to rename the filename with a PR on cmsis-svd-data and keep your code as it, else fix the indexer and downloader.
```txt
NXP/LPC11Axxv0.6.svd       ->    NXP.LPC11Axxv0
NXP/LPC11E6x_v0.8.svd      ->    NXP.LPC11E6x_v0
NXP/LPC15xx_v0.7.svd       ->    NXP.LPC15xx_v0
NXP/LPC176x5x_v0.2.svd     ->    NXP.LPC176x5x_v0
NXP/LPC178x_7x_v0.8.svd    ->    NXP.LPC178x_7x_v0
NXP/LPC408x_7x_v0.7.svd    ->    NXP.LPC408x_7x_v0
NXP/LPC5410x_v0.4.svd      ->    NXP.LPC5410x_v0
NXP/LPC800_v0.3.svd        ->    NXP.LPC800_v0
```

VincentDary, 2024-10-22:
@brainstorm @VincentDary Any feedback on these two implementations?

@BenBE of course :)

Thank you very much for your great proposal an this big POC improvement, below my code review for svd_dl.py:

- line [33](https://github.com/BenBE/cmsis-svd/blob/svd-downloader/python/cmsis_svd/svd_dl.py#L33): remove `List` since it is not used.
- line [35](https://github.com/BenBE/cmsis-svd/blob/svd-downloader/python/cmsis_svd/svd_dl.py#L35): replace `URLError` by `HTTPError`.
- line [20](https://github.com/BenBE/cmsis-svd/blob/svd-downloader/python/cmsis_svd/svd_dl.py#L20), [21](https://github.com/BenBE/cmsis-svd/blob/svd-downloader/python/cmsis_svd/svd_dl.py#L21): remove asyncio related import since there are not used.

```diff
@@ -17,8 +17,6 @@

 import argparse
 import asyncio
-import contextvars
-import functools
 import gzip
 import hashlib
 import json
@@ -30,9 +28,9 @@ import socket
 import tarfile
 import time

-from typing import Dict, List, Optional, Union
+from typing import Dict, Optional, Union
 from urllib import request
-from urllib.error import URLError
+from urllib.error import HTTPError
```

- line [71](https://github.com/BenBE/cmsis-svd/blob/svd-downloader/python/cmsis_svd/svd_dl.py#L71), [367](https://github.com/BenBE/cmsis-svd/blob/svd-downloader/python/cmsis_svd/svd_dl.py#L367), [489](https://github.com/BenBE/cmsis-svd/blob/svd-downloader/python/cmsis_svd/svd_dl.py#L489): remove asyncio related code since it is not used, file download are inlined in your improvements.

```diff
@@ -68,13 +66,6 @@ LOCAL_DATA_INDEX_JSON_ZSTD = os.path.join(LOCAL_DATA_DIR, INDEX_JSON+'.zstd')
 LOCAL_DATA_INDEX_HASH = os.path.join(LOCAL_DATA_DIR, INDEX_HASH)

-async def to_thread(func, /, *args, **kwargs):
-    loop = asyncio.get_running_loop()
-    ctx = contextvars.copy_context()
-    func_call = functools.partial(ctx.run, func, *args, **kwargs)
-    return await loop.run_in_executor(None, func_call)
-
-
 class SvdDlError(Exception):
     pass
```
.

```diff
@@ -364,7 +354,7 @@ class SvdDl:

         pass

-    async def download_svd(self, download_string: str) -> None:
+    def download_svd(self, download_string: str) -> None:
         download_files, download_packs = [], []
         os.makedirs(LOCAL_DATA_DIR, exist_ok=True)
```
.
```diff
@@ -486,7 +475,7 @@ def main():
     script_args = parser.parse_args()

     if script_args.svd_dl_subparser == sub_parser_dl:
-        asyncio.run(SvdDl(repo=script_args.svd_repo).download_svd(script_args.svd_resources))
+        SvdDl(repo=script_args.svd_repo).download_svd(script_args.svd_resources)

 if __name__ == '__main__':
     main()
[... 123 lines cut ...]
         if index_json_hash != index_json_hash_valid:
-            raise SvdDlError(f'"index.json" is corrupted.')
+            raise SvdDlError(f'"{INDEX_JSON}" is corrupted.')

         with open(LOCAL_DATA_INDEX_JSON, 'r') as f:
             self.index_json = json.load(f)

         if not self._validate_index():
-            raise SvdDlError(f'"index.json" contains invalid data.')
+            raise SvdDlError(f'"{INDEX_JSON}" contains invalid data.')

```

- line [437](https://github.com/BenBE/cmsis-svd/blob/svd-downloader/python/cmsis_svd/svd_dl.py#L437): `download_files` can be empty since all the packages contain all the files.

```diff
@@ -420,21 +410,20 @@ class SvdDl:

         if download_string == 'ALL':
             download_packs = self.index_json['packages'].keys()
-            download_files = self.index_json['files'].keys()
         else:
             svd_resources = download_string.split(',')
             pkg_names = self.index_json['packages'].keys()
```

BenBE, 2024-10-23:
> * Maybe add a `dependency.txt` file at the root of the repository with a `pyzstd` library reference and a frozen version.

This file should be replicated (or at least its contents) for both `cmsis-svd` and `cmsis-svd-data`, because both the indexer and the downloader have this optional dependency.

> * line [101](https://github.com/BenBE/cmsis-svd-data/blob/svd-indexer/indexer.py#L101): There is a bug at this line because some SVD filename are using dotted version number :/, see below, so IMO [...]

Good catch. I think both is fine. First off, the dotted name should be opaque for the user, thus no calculations should be performed using it to derive anything. Thus the indexer code here is wrong in the sense, that it should operate on the split path and derive the correct pack name from that. Renaming SVD files should be the exception as it may break backwards compatibility with existing installations that expect the old (file) name. Will fix that part later when I do some refactoring of that part.

Will push the updated branch once I'm done with updating things in both the indexer and downloader.