cmsis-svd/cmsis-svd #193: CMSIS-SVD.xsd: type "dimableIdentifierType" for attribute "derivedFrom" does not allow dot ('.')

andbararm, 2024-10-23:
In the current CMSIS-SVD.xsd 1.3.9, the type "dimableIdentifierType" is used for the attribute "derivedFrom". "derivedFrom" may contain dots to specify the hierarchy/full qualifying path of the referenced object. However, the type "dimableIdentifierType" does not allow dots in the string. And so checking the svd containing things like:

`<field derivedFrom="Peripheral_1.Register_1_1.Field_1_1_1">
`

 against the CMSIS-SVD.xsd will produce errors like this:

`Cvc-pattern-valid: Value 'Peripheral_1.Register_1_1.Field_1_1_1' Is Not Facet-valid With Respect To Pattern '((%s)|(%s)[_A-Za-z]{1}[_A-Za-z0-9]*)|([_A-Za-z]{1}[_A-Za-z0-9]*(\[%s\])?)|([_A-Za-z]{1}[_A-Za-z0-9]*(%s)?[_A-Za-z0-9]*)' For Type 'dimableIdentifierType'., Line '74', Column '72'.`

VincentDary, 2024-10-25:
This is a know issue with the schema 1.3.9, @ckudera have reported this problem to the maintainers of the CMSIS SVD spec, see ["Fix derivedFrom attribute types and enumeratedValue minOccurs"](https://github.com/Open-CMSIS-Pack/svd-spec/pull/14).

The @ckudera's CMSIS SVD spec patch actually in review by the CMSIS SVD spec maintainers is available in this repository, see [schemas/CMSIS-SVD_1_3_11.xsd](https://github.com/cmsis-svd/cmsis-svd/blob/main/schemas/CMSIS-SVD_1_3_11.xsd#L490)

To explicitly use the SVD schema 1.3.11 with Python cmsis-svd, set `xml_validation` to True, `schema_version_detection` to `False` and `schema_version` to `1.3.11`.

```python
from cmsis_svd import SVDParser
parser = SVDParser.for_xml_file('your_svd_file.svd')
svd_device = parser.get_device(xml_validation=True, schema_version='1.3.11', schema_version_detection=False)
```

Or without `schema_version` parameter since the latest version of the CMSIS SVD schema is used if not specified.

```python
svd_device = parser.get_device(xml_validation=True, schema_version_detection=False)
```

VincentDary, 2024-10-25:
@andbararm the version 1.3.11 is now the last official version of the CMSIS SVD spec, since 1 hour.

closed 2024-10-25 by VincentDary